rainchi/ESPHome-LD2410 #2: Documentation on how to use and program.

imoldovavan, 2022-11-02:
I successfully connected the sensor to and esp8266. I am a little confused on how to program and test this. The way I think programming works is like this.
1. To program Turn on config mode
2. Adjust the sliders for max distances and adjust the None Duration number (is this in seconds??). Values are saved to the sensor every time they are changed. 
3. Turn off config mode to return to normal operation.
4. Get config only works when Config mode is on??

The sensor over a period of 24 hours has a lot of false positives in the log.  I am still playing with values to see if this can be improved.
I connected the sensor to this switch like the one pictured here https://github.com/imoldovavan/North-WI-FI-Outlet-ESPHome-hack . I will have a complete how to do if I can figure out how to reduce or eliminate the false positives. Thank you for creating this, and thank you for your help.

pioneershahid, 2022-11-07:
This might help as there is a PR currently in-progress.
https://github.com/esphome/esphome/pull/3919 and https://github.com/esphome/esphome-docs/pull/2385

imoldovavan, 2022-11-10:
That is awesome. Can't wait for the release. Thank you...

closed 2022-11-10 by imoldovavan