rainchi/ESPHome-LD2410 #4: Error while compiling - ld2410_uart.h

softhuntermike, 2022-11-08:
Got so many error during compiling, I copied/pasted here below the log. 
ld2410_uart.h file

```
Compiling /data/ld2410-esp01-01/.pioenvs/ld2410-esp01-01/src/main.cpp.o
In file included from src/main.cpp:85:
src/ld2410_uart.h:219:10: warning: missing terminating ' character
  219 |     <!-- '"` --><!-- </textarea></xmp> --></option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="520147280" data-scoped-search-url="/rain931215/ESPHome-LD2410/search" data-owner-scoped-search-url="/users/rain931215/search" data-unscoped-search-url="/search" data-turbo="false" action="/rain931215/ESPHome-LD2410/search" accept-charset="UTF-8" method="get">
      |          ^
src/ld2410_uart.h:219:10: error: missing terminating ' character
  219 |     <!-- '"` --><!-- </textarea></xmp> --></option></form><form class="js-site-search-form" role="search" aria-label="Site" data-scope-type="Repository" data-scope-id="520147280" data-scoped-search-url="/rain931215/ESPHome-LD2410/search" data-owner-scoped-search-url="/users/rain931215/search" data-unscoped-search-url="/search" data-turbo="false" action="/rain931215/ESPHome-LD2410/search" accept-charset="UTF-8" method="get">
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/ld2410_uart.h:276:75: error: extended character ↵ is not valid in an identifier
  276 |       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
      |                                                                           ^
src/ld2410_uart.h:281:56: error: extended character ↵ is not valid in an identifier
  281 |       <span class="d-inline-block ml-1 v-align-middle">↵</span>
      |                                                        ^
src/ld2410_uart.h:323:75: error: extended character ↵ is not valid in an identifier
  323 |       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
      |                                                                           ^
src/ld2410_uart.h:328:56: error: extended character ↵ is not valid in an identifier
  328 |       <span class="d-inline-block ml-1 v-align-middle">↵</span>
      |                                                        ^
src/ld2410_uart.h:361:75: error: extended character ↵ is not valid in an identifier
  361 |       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
      |                                                                           ^
src/ld2410_uart.h:366:56: error: extended character ↵ is not valid in an identifier
  366 |       <span class="d-inline-block ml-1 v-align-middle">↵</span>
      |                                                        ^
src/ld2410_uart.h:399:75: error: extended character ↵ is not valid in an identifier
  399 |       <span aria-hidden="true" class="d-inline-block ml-1 v-align-middle">↵</span>
      |                                                                           ^
src/ld2410_uart.h:404:56: error: extended character ↵ is not valid in an identifier
  404 |       <span class="d-inline-block ml-1 v-align-middle">↵</span>
      |                                                        ^
In file included from src/main.cpp:85:
src/ld2410_uart.h:447:10: warning: missing terminating ' character
  447 |     <!-- '"` --><!-- </textarea></xmp> --></option></form><form data-turbo="false" action="/logout" accept-charset="UTF-8" method="post"><input type="hidden" name="authenticity_token" value="pdTaeIjUmiG-A80v_sZ8XS3dPfCAW4QhlJahUjI-6pE0SB7T-3lARCUhrTNKzdYpBAmDxTAkjQpGmu8FeXL8Eg" />
      |          ^
src/ld2410_uart.h:447:10: error: missing terminating ' character
  447 |     <!-- '"` --><!-- </textarea></xmp> --></option></form><form data-turbo="false" action="/logout" accept-charset="UTF-8" method="post"><input type="hidden" name="authenticity_token" value="pdTaeIjUmiG-A80v_sZ8XS3dPfCAW4QhlJahUjI-6pE0SB7T-3lARCUhrTNKzdYpBAmDxTAkjQpGmu8FeXL8Eg" />
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/main.cpp:85:
src/ld2410_uart.h:640:19: warning: missing terminating " character
  640 |       data-action="
      |                   ^
src/ld2410_uart.h:640:19: error: missing terminating " character
src/ld2410_uart.h:641:52: error: stray '#' in program
  641 |         command-palette-input-ready:command-palette#inputReady
      |                                                    ^
src/ld2410_uart.h:642:59: error: stray '#' in program
  642 |         command-palette-page-stack-updated:command-palette#updateInputScope
      |                                                           ^
src/ld2410_uart.h:643:37: error: stray '#' in program
  643 |         itemsUpdated:command-palette#itemsUpdated
      |                                     ^
src/ld2410_uart.h:644:32: error: stray '#' in program
  644 |         keydown:command-palette#onKeydown
      |                                ^
src/ld2410_uart.h:645:44: error: stray '#' in program
  645 |         loadingStateChanged:command-palette#loadingStateChanged
      |                                            ^
src/ld2410_uart.h:646:44: error: stray '#' in program
  646 |         selectedItemChanged:command-palette#selectedItemChanged
      |                                            ^
src/ld2410_uart.h:647:39: error: stray '#' in program
  647 |         pageFetchError:command-palette#pageFetchError
      |                                       ^
src/ld2410_uart.h:648:7: warning: missing terminating " character
[... 276 lines cut ...]
      |                                                                                                                                                                                                   ^
src/ld2410_uart.h:2810:193: error: expected constructor, destructor, or type conversion before ';' token
 2810 |           <td id="LC3" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">define</span> <span class="pl-en">CHECK_BIT</span>(<span class="pl-v">var, pos</span>) (((var) &gt;&gt; (pos)) &amp; <span class="pl-c1">1</span>)</td>
      |                                                                                                                                                                                                 ^
src/ld2410_uart.h:2810:200: error: expected constructor, destructor, or type conversion before ')' token
 2810 |           <td id="LC3" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">define</span> <span class="pl-en">CHECK_BIT</span>(<span class="pl-v">var, pos</span>) (((var) &gt;&gt; (pos)) &amp; <span class="pl-c1">1</span>)</td>
      |                                                                                                                                                                                                        ^
src/ld2410_uart.h:2810:208: error: expected unqualified-id before '<' token
 2810 |           <td id="LC3" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">define</span> <span class="pl-en">CHECK_BIT</span>(<span class="pl-v">var, pos</span>) (((var) &gt;&gt; (pos)) &amp; <span class="pl-c1">1</span>)</td>
      |                                                                                                                                                                                                                ^
In file included from src/main.cpp:85:
src/ld2410_uart.h:3941:75: error: expected unqualified-id before '<' token
 3941 |           <td id="LC280" class="blob-code blob-code-inner js-file-line">};</td>
      |                                                                           ^
In file included from src/main.cpp:85:
src/ld2410_uart.h:4018:16: error: expected unqualified-id before numeric constant
 4018 |         &copy; 2022 GitHub, Inc.
      |                ^~~~
In file included from src/main.cpp:85:
src/ld2410_uart.h:4111:5: error: expected unqualified-id before '<' token
 4111 |     </style>
      |     ^
*** [/data/ld2410-esp01-01/.pioenvs/ld2410-esp01-01/src/main.cpp.o] Error 1
======================== [FAILED] Took 1298.76 seconds ========================
```

lienbacher, 2022-11-08:
seema like you have html code in your header file. that wont make a c compiler happy.

softhuntermike, 2022-11-09:
I downloaded the header file without making any modifications. Anyway I will check again.

softhuntermike, 2022-11-09:
It works now. I open, copy pasted text from ld2410_uart.h file into a text editor and saved a new file.

closed 2023-02-20 by rain931215