rainchi/ESPHome-LD2410 #5: All sensors are "Unknown" after compilation and upload

miljume, 2022-12-02:
All sensors become "Unknown" (Okänd = Unknown) after startup
[image]

rain931215, 2022-12-07:
Can you share your wiring for ld2410?

miljume, 2022-12-07:
ESP8266 | LD2410
5V <-> VCC
GND <-> GND
TX <-> RX
RX <-> TX

cmille34, 2022-12-08:
I am in the same boat with a D1 mini.  I don't see any uart data when viewing the logs remotely with debugging on. The output pin seems to be working though.

Im powering the LD2410 from the 5v pin on the D1 mini, i tried 5v directly from a power supply as well to the sensor and that didn't make a difference.  Any other ideas why  the sensors show unknown?

Below is my YAML as well in case it helps.

`esphome:
  name: chrisofficemmwave
  includes:
    - ld2410_uart.h
  on_boot:
    priority: 600
    # ...
    then:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setNumbers(maxMovingDistanceRange, maxStillDistanceRange, noneDuration);	

esp8266:
  board: d1_mini
  framework:
    version: latest

logger:
  level: DEBUG  #You Can Use "INFO" Level
  baud_rate: 0
api:
ota:
web_server:
  port: 80  
wifi:
  ssid: ****
  password: ***

uart:
  id: uart1
  rx_pin: GPIO3  #ESP8266 UART_0
  tx_pin: GPIO1  #ESP8266 UART_0
  baud_rate: 256000
  data_bits: 8
  stop_bits: 1
  parity: NONE
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
      
custom_component:
  - lambda: |-
      return {new LD2410(id(uart1))};
    components:
      - id: ld2410
      
binary_sensor:
  - platform: custom
    lambda: |-
      auto uart_component = static_cast<LD2410 *>(ld2410);
      return {uart_component->hasTarget,uart_component->hasMovingTarget,uart_component->hasStillTarget,uart_component->lastCommandSuccess};
    binary_sensors:
      - name: "Has Target"
      - name: "Has Moving Target"
      - name: "Has Still Target"
      - name: "Last Command Success"
  - platform: gpio
    id: MovOcc_gpio  
    pin:
      number: D4
      mode:
        input: true
        pullup: true    
    name: "mmWave Pin"
    device_class: occupancy        
  
sensor:
  - platform: custom
    lambda: |-
      auto uart_component = static_cast<LD2410 *>(ld2410);
      return {uart_component->movingTargetDistance,uart_component->movingTargetEnergy,uart_component->stillTargetDistance,uart_component->stillTargetEnergy,uart_component->detectDistance};
    sensors:
      - name: "Moving Target Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0
      - name: "Moving Target Energy"
        unit_of_measurement: "%"
        accuracy_decimals: 0
      - name: "Still Target Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0
      - name: "Still Target Energy"
        unit_of_measurement: "%"
        accuracy_decimals: 0
      - name: "Detect Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0

number:        
  - platform: template
    name: "Max Moving Distance Range"
    id: maxMovingDistanceRange
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(x,id(maxStillDistanceRange).state,id(noneDuration).state);
  - platform: template
    name: "Max Still Distance Range"
    id: maxStillDistanceRange
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(id(maxMovingDistanceRange).state,x,id(noneDuration).state);
  - platform: template
    name: "None Duration"
    id: noneDuration
    min_value: 0
    max_value: 32767
    step: 1
    mode: box
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(id(maxMovingDistanceRange).state,id(maxStillDistanceRange).state,x);
      
button:
  - platform: template
    name: "Reboot LD2410"
    on_press:
      lambda: 'static_cast<LD2410 *>(ld2410)->reboot();'
  - platform: template
    name: "Turn on config mode"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setConfigMode(true);'
  - platform: template
    name: "Turn off config mode"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setConfigMode(false);'
  - platform: template
    name: "Get config"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->queryParameters();'
  - platform: template
    name: "Set baud rate to 256000"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(7);'
  - platform: template
    name: "Set baud rate to 115200"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(5);'
  - platform: template
    name: "Set baud rate to 9600"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(1);'`

cmille34, 2022-12-08:
I swapped over to using RX = GPIO13/D7  and TX=GPIO15/D8 on the D1 min and now I am getting uart data.  Leaving this here in case it helps someone else.

n1kDon, 2022-12-19:
@cmille34 Wow that worked!!! Bought 5 more after the first one worked flawless. There was no uart data on all of the new ones. BIG thanks, I dont know why but now it works! I've been messing around with these things for 2 days. 

@rain931215 maybe you should point this out in the docu

miljume, 2022-12-20:
> I swapped over to using RX = GPIO13/D7 and TX=GPIO15/D8 on the D1 mini and now I am getting uart data. Leaving this here in case it helps someone else.

Can you please attach both your .yaml file and your .h include?
Cant get it to compile

cmille34, 2022-12-20:
> > I swapped over to using RX = GPIO13/D7 and TX=GPIO15/D8 on the D1 mini and now I am getting uart data. Leaving this here in case it helps someone else.
> 
> @cmille34: Can you please attach both your .yaml file and your .h include? Cant get it to compile

Here is my full yaml.  My .h file is the one from this repository. I didn't mess with that.

`esphome:
  name: chrisofficemmwave
  includes:
    - ld2410_uart.h
  on_boot:
    priority: 600
    # ...
    then:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setNumbers(maxMovingDistanceRange, maxStillDistanceRange, noneDuration);	

esp8266:
  board: d1_mini
  framework:
    version: latest

logger:
  level: DEBUG  #You Can Use "INFO" Level
  baud_rate: 0
api:
ota:
web_server:
  port: 80  
wifi:
  ssid: **
  password: **

uart:
  id: uart1
  rx_pin: GPIO13  #ESP8266 UART_0
  tx_pin: GPIO15  #ESP8266 UART_0
  baud_rate: 256000
  data_bits: 8
  stop_bits: 1
  parity: NONE
#  debug:
#    direction: BOTH
#    dummy_receiver: false
#    after:
#      delimiter: "\n"
#    sequence:
#      - lambda: UARTDebug::log_string(direction, bytes);
      
custom_component:
  - lambda: |-
      return {new LD2410(id(uart1))};
    components:
      - id: ld2410
      
binary_sensor:
  - platform: custom
    lambda: |-
      auto uart_component = static_cast<LD2410 *>(ld2410);
      return {uart_component->hasTarget,uart_component->hasMovingTarget,uart_component->hasStillTarget,uart_component->lastCommandSuccess};
    binary_sensors:
      - name: "Has Target"
      - name: "Has Moving Target"
      - name: "Has Still Target"
      - name: "Last Command Success"
  - platform: gpio
    id: MovOcc_gpio  
    pin:
      number: GPIO14
      mode:
        input: true
        pullup: true    
    name: "mmWave Pin"
    filters:
      - delayed_on: 10ms    
    device_class: occupancy        
  
sensor:
  - platform: custom
    lambda: |-
      auto uart_component = static_cast<LD2410 *>(ld2410);
      return {uart_component->movingTargetDistance,uart_component->movingTargetEnergy,uart_component->stillTargetDistance,uart_component->stillTargetEnergy,uart_component->detectDistance};
    sensors:
      - name: "Moving Target Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0
      - name: "Moving Target Energy"
        unit_of_measurement: "%"
        accuracy_decimals: 0
      - name: "Still Target Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0
      - name: "Still Target Energy"
        unit_of_measurement: "%"
        accuracy_decimals: 0
      - name: "Detect Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0

number:        
  - platform: template
    name: "Max Moving Distance Range"
    id: maxMovingDistanceRange
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(x,id(maxStillDistanceRange).state,id(noneDuration).state);
  - platform: template
    name: "Max Still Distance Range"
    id: maxStillDistanceRange
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(id(maxMovingDistanceRange).state,x,id(noneDuration).state);
  - platform: template
    name: "None Duration"
    id: noneDuration
    min_value: 0
    max_value: 32767
    step: 1
    mode: box
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(id(maxMovingDistanceRange).state,id(maxStillDistanceRange).state,x);
      
button:
  - platform: template
    name: "Reboot LD2410"
    on_press:
      lambda: 'static_cast<LD2410 *>(ld2410)->reboot();'
  - platform: template
    name: "Turn on config mode"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setConfigMode(true);'
  - platform: template
    name: "Turn off config mode"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setConfigMode(false);'
  - platform: template
    name: "Get config"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->queryParameters();'
  - platform: template
    name: "Set baud rate to 256000"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(7);'
  - platform: template
    name: "Set baud rate to 115200"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(5);'
  - platform: template
    name: "Set baud rate to 9600"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(1);'`

miljume, 2022-12-20:
@cmille34 Worked perfect, thanks a lot!

closed 2022-12-20 by miljume