rainchi/ESPHome-LD2410 #8: Unknown sensor values

southalladam, 2023-01-31:
Firstly, thanks for writing this integration. 

I have copied the header file and updated my config based on your supplied config. Everything seems to have worked when building and installing my latest firmware, but the sensor values are 'unknown'.

This is a new device, not previously configured with the HiLink tool, which I assume is ok as this is not listed as a step on your readme page.

I set my max moving distance and max still distance, but insure about how I should be using the config mode.

This is probably not an issue with the code, more likely some steps I am not aware of. Happy to provide some updates to readme once I understand what needs to be done.

Many thanks
Adam

rain931215, 2023-02-20:
It seems that LD2410 will freeze up randomly when using config mode because  it's high UART baudrate.
And this might cause all sensors become unknown.

southalladam, 2023-02-20:
Thanks for the response.

In this case it was a problem with my wiring. I switched my connection from TX/RX to two alternative GPIOs and it worked like a charm! 

Not sure if it was simply a bad connection or maybe I shouldn't be using TX/RX pins for this?

closed 2023-02-20 by southalladam