rainchi/ESPHome-LD2410 #14: Not working with newer version of esp8266

Trigun, 2023-03-25:
Hi, it seems that the Ld2410 in combination with this code does not seem to work work with newer versions of esp8266. I soldered three. 1 with an older version of the esp which works great! Two with a newer(different) version which results in all entities being unknown in Home assistant. Is there something that needs to be changed in the code?

Thanks a lot!

tientmit88, 2023-03-26:
[image]
me too, it doesn't work

captshadab, 2023-03-28:
Can I soldier it to esp32?

krisalianne, 2023-04-02:
Same thing here. Doesn't work anymore.

captshadab, 2023-04-02:
> Same thing here. Doesn't work anymore.

Hey 

I soldered the esp32 with ld2410b and it works however the yaml is slightly different

krisalianne, 2023-04-02:
Hi,
Thanks for your reply, could you share your config file, please?

captshadab, 2023-04-03:
hi here you go

`substitutions:
  name: esp32-bluetooth-proxy-4ca418
  friendly_name: xxxxxxxxx
  

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

ota:  

api:
  encryption:
    key: 1x7H+ykUVBotcXZ+ySbimRktvLAfGm5KW+Sk=

wifi:
  networks:
  - ssid: "xxxxxx"
    password: "xxxxxxx"
    channel: X  
  output_power: 19.5db
  fast_connect: True
  power_save_mode: light
  manual_ip:
    static_ip: 192.168.0.91
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: False

bluetooth_proxy:
  active: True

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended 

switch:
  - platform: restart
    name: xxxxxxx restart  

external_components:
  - source: github://esphome/esphome@dev
    components: [ ld2410 ]

logger:
  level: DEBUG
  baud_rate: 0

uart:
  id: uart1
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  timeout: 10s
  max_move_distance : 6m
  max_still_distance: 6m
  g0_move_threshold: 50
  g0_still_threshold: 20
  g1_move_threshold: 50
  g1_still_threshold: 20
  g2_move_threshold: 40
  g2_still_threshold: 20
  g3_move_threshold: 30
  g3_still_threshold: 20
  g4_move_threshold: 30
  g4_still_threshold: 20
  g5_move_threshold: 20
  g5_still_threshold: 20
  g6_move_threshold: 20
  g6_still_threshold: 20
  g7_move_threshold: 40
  g7_still_threshold: 30
  g8_move_threshold: 40
  g8_still_threshold: 30

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target

sensor:
  - platform: ld2410
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance  

`

krisalianne, 2023-04-03:
Thank You very much, I'll give it a try today.

krisalianne, 2023-04-04:
It works perfectly, thanks again.

cmussoni, 2023-04-04:
Hi, I'm also interested in this: what do you mean by "newer(different)versions" of esp8266 ?  Thank you. Claudio.

Trigun, 2023-04-04:
[image]

For example. In this image there are two esp8266 boards. The right one works perfectly with the LD2410b. The left one doesnt seems to work. The output gives all the sensors an unknown state. Same code only a different board

krisalianne, 2023-04-04:
You're right, just test it with the board on the right, it work.

systemerror-codenotfound, 2023-05-05:
> [image]
> 
> For example. In this image there are two esp8266 boards. The right one works perfectly with the LD2410b. The left one doesnt seems to work. The output gives all the sensors an unknown state. Same code only a different board

i only have the left board, any idea about how i can use it?

closed 2026-11-06 by Trigun