rainchi/ESPHome-LD2410 #15: LD2410B option to disable/enable bluetooth

joaopedros2, 2023-04-02:
Hello!
Is it possible to have the option to turn on and off the sensor's Bluetooth?
I have model B and I'm not using it.
I saw in another project that it allows control:
https://github.com/screek-workshop/screek-human-sensor/tree/main/1w/yaml/sun-light-reader/screek-mod-components/ld2410

Thanks 🙏

lienbacher, 2023-04-24:
There is an update for the official integration in esphome in the making/available that has this and many other improvements. I don't know if it has been merged yet.

joaopedros2, 2023-04-25:
Can you share the official integration?

lienbacher, 2023-04-25:
Here's the official one: https://esphome.io/components/sensor/ld2410.html?highlight=ld2410
and this is the PR I was referencing: https://github.com/esphome/esphome/pull/4434
You should be able to use the example config with the current release of esphome.