rainchi/ESPHome-LD2410 #19: old config is failing to compile/update

FGOD1983, 2023-08-17:
it seems that there has been changes made to the way the LD2410 is being configured in ESP Home and therefor my old config is not valid anymore. I tried changing them by looking at examples here and the documentation on the esp page which states that the use of numbers is now needed, but somehow I am unable to get the config working

old config errors:
[image]

the new config is being accepted, but fails to compile with tons of errors (which is basically a config taken from one of the issues here stating it should work).

my old config:
```
esphome:
  name: naambord-esp8266
  friendly_name: Naambord-ESP8266

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "api-key"  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: x.x.x.x
    gateway: x.x.x.x
    subnet: x.x.x.x
    dns1: x.x.x.x
    dns2: x.x.x.x

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Naambord-ESP8266"
    password: "wifi-ap-pass"

captive_portal:
    
ota:
  password: "ota-pass"

# LED strip
light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812X
    pin: GPIO2
    num_leds: 34
    name: "NeoPixel Light"

# LD2410 sensor
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  timeout: 150s
  max_move_distance : 2.25m
  max_still_distance: 0.75m
  #g0_move_threshold: 10
  #g0_still_threshold: 20
  #g1_move_threshold: 10
  #g1_still_threshold: 20
  #g2_move_threshold: 20
  #g2_still_threshold: 21
  g3_move_threshold: 30
  #g3_still_threshold: 31
  #g4_move_threshold: 40
  #g4_still_threshold: 41
  #g5_move_threshold: 50
  g5_still_threshold: 46
  #g5_still_threshold: 51
  #g6_move_threshold: 60
  #g6_still_threshold: 61
  #g7_move_threshold: 70
  #g7_still_threshold: 71
  #g8_move_threshold: 80
  #g8_still_threshold: 81

sensor:
  - platform: ld2410
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
```

lienbacher, 2023-08-17:
While this is not a solution for this specific problem, but in the meanwhile esphome has released an official component for the ld2410 and have also just merged a significant update into the dev branch that improves configuration flow abilities. 

I personally would consider this repository obsolete at this point, there have not been any updates to this for more than a year by now.

FGOD1983, 2023-08-17:
seems so indeed, and although the docs of the updated official component is far from ideal, it seems it works if you change the whole config to the number components with the official ld2410 component but leave the values open. it then let's you set those values in HA, but as I was unable to get that working I kinda got issues getting any version to work. 

my now working config with the official component is as follows (for whoever has this issue, btw, this also holds a setup for a ledstrip connnected to this same D1 mini):

```
esphome:
  name: naambord-esp8266
  friendly_name: Naambord-ESP8266

esp8266:
  board: esp01_1m

external_components:
  source: github://regevbr/esphome@ld2410
  components: [ld2410]

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "api-key"  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: x.x.x.x
    gateway: x.x.x.x
    subnet: x.x.x.x
    dns1: x.x.x.x
    dns2: x.x.x.x

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Naambord-ESP8266"
    password: "ap-pass"

captive_portal:
    
ota:
  password: "ota-pass"

# LED strip
light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812X
    pin: GPIO2
    num_leds: 34
    name: "NeoPixel Light"

# LD2410 sensor
uart:
  id: ld2410_uart
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  uart_id: ld2410_uart
  throttle: 1500ms
  id: ld2410_comp

select:
  - platform: ld2410
    distance_resolution:
      name: distance resolution
    baud_rate:
      name: baud rate
    light_function:
      name: light function
    out_pin_level:
      name: out pin level

button:
  - platform: ld2410
    factory_reset:
      name: "factory reset"
    restart:
      name: "restart"
    query_params:
      name: query params

number:
  - platform: ld2410
    timeout:
      name: timeout
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold
    light_threshold:
      name: light threshold

text_sensor:
  - platform: ld2410
    version:
      name: "presenece sensor version" 
    mac_address:
      name: "presenece sensor mac address"

switch:
  - platform: ld2410
    engineering_mode:
      name: "engineering mode" 

binary_sensor:
  - platform: ld2410
    has_target:
      name: "Presence"
    has_moving_target:
      name: "Moving Target"
    has_still_target:
      name: "Still Target"
    out_pin_presence_status:
      name: out pin presence status
  - platform: gpio
    pin: GPIO1
    name: gpio out pin presence
    device_class: presence

sensor:
  - platform: ld2410
    moving_distance:
      name: "Moving distance (cm)"
    still_distance:
      name: "Still Distance (cm)"
    moving_energy:
      name: "Move Energy (%)"
    still_energy:
      name: "Still Energy (%)"
    detection_distance:
      name: "Distance Detection (cm)"
    g0: 
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1: 
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2: 
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3: 
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4: 
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5: 
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6: 
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7: 
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8: 
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy
    light:
      name: light
```