rainchi/ESPHome-LD2410 #22: ESPHome has removed the custom_component

Vesa62, 2025-03-06:
ESPHome has removed the custom_component feature and now recommends using external components. This means you will need to create an external component for the LD2410.

makerwolf, 2025-03-06:
this component is long obsolete because ESPhome ships with a tried and tested component for the LD2410 since a rather long time now ... https://esphome.io/components/sensor/ld2410.html

Vesa62, 2025-03-07:
Thanks, makerwolf for your reply. I am a rookie with esp:s and github. 
I found the article you have linked to but I could not get any sense with that. Eventually I found complete code (https://www.simplysmart.house/blog/presence-detection-ld2410-home-assistant#choosing-esp32-over-bluetooth), which worked with a few mods.

I need to learn how to get the device working by using the document you linked. All tips are appreciated.